BorisBrock/HeidelBridge #2: Modbus Connection Issue with Heidelbridge - Wallbox Flashing White and Blue

darkblum, 2024-10-01:
Unfortunately, I am unable to establish a connection through the Heidelbridge.

I followed your instructions carefully:  
- Set S4/4 and S6/2 bit 1 as instructed  
- Verified the pin connections  
- Changed the WLAN Settings and flashed as instructed

However, the Wallbox is still flashing white and blue, indicating a Modbus error.

I also added a 120-ohm resistor between Pin A and B, but there has been no change.

Did I overlook something?

---

Let me know

BorisBrock, 2024-10-01:
Hi darkblum,
there are several things to check:

1. Please double check, that the Heidelberg wallbox is configured correctly. See the [official manual here ](https://green-energy-shop.com/images/Datenblatt/Heidelberger_3_Wallbox_Energy_Control_Externes_Lastmanagement.pdf)
2. Make sure the connection between the wallbox and your rs485 board is correct (wires not crossed? all wires firmly secured? 120 Ohm resistor? Shielding only connected at one end (wallbox)?).
3. Check your ESP32+RS485 setup: are the modules properly interconnected? Did you use the correct pins?

Please also check the ESP32 serial port outputs. It should print a lot of messages (especially on startup), that might be helpful.

Also: what kind of setup are you using? Breadboard or custom PCB? Which ESP32 board are you using exactly?

darkblum, 2024-10-02:
Hello BorisBrock,

I've managed to establish the correct connection to the wallbox. Initially, I thought that the A/B connection on the left and right sides of the PCB had the same orientation, but that wasn’t the case. On the left side, it’s B/A, while on the right side, it’s A/B. So now, there's no more blinking.

I’ve also integrated the charger into EVCC and set up the load point, but my EV isn’t registering as "connected." The vehicle status always shows as "disconnected," and I’m getting the information through MQTT. My EV is entering a charging failure state due to no current being supplied.

I am using ESP32 WROOM-32D with usb-c connector and no breadboard only jumper cables and WAGO connectors.

What should I check next?

BorisBrock, 2024-10-02:
Hi darkblum,
good to hear that the Modbus connection is now working :+1: 

I'm using this project with a VW ID3 and the vehicle connection detection works 100% reliably. This week I also connected a Skoda Enyaq from a neighbor and that also worked very well. Which car are you using?

For solving your issue, I would recommend reading the serial outputs from the ESP32 with a serial monitor. HeidelBridge will read the register containing the connection status cyclically every few seconds, and there should be plenty of messages. This way we can see, if the wallbox is not detecting your vehicle at all, or if reading the register fails.

So:
 - Connect the car
 - Wait a few seconds
 - Start logging the serial outputs for a minute or so
 - Post the results here

BorisBrock, 2024-10-02:
Hi, it should not make any difference which car you are using. Only if you want it to show up in evcc in a defined way, you would need to do some additional configuration there.
But regarding HeidelBridge/Home Assistant there is nothing to do here.

The VS Code Serial Monitor should work fine. That's what I'm using as well.

PS: When compiling the code you can choose between two environments: "dummy" and "heidelberg". You could also try using "dummy", which will provide some hard coded values. These should definitely show up in MQTT/Home Assistant.
For communicating with the real wallbox, make sure to compile the "heidelberg" environment though.

nobsch, 2024-10-30:
Hello, first of all thank you for the great project. Unfortunately my Heidelberg EC is also blinking the Modbus Failure. It starts when configuring the Wallbox for Modbus and never ends. I checked the DIP Switches several timesI tried different rs232-Modbus Converter and also the wiring. No change.
The crazy thing is, that the HeidelBridge can read the Wallbox (Temperature, Power) and write the charge current. So in fact the communication should be OK. But the Wallbox is continously blinking and the connection to the car is not recognized. No charging. When I change the configuration of the Wallbox (S4,S6) all is fine.
Do you have a hint for me?

BorisBrock, 2024-10-31:
First of all you should check what kind of error you are getting EXACTLY. The different blink patterns are described in the [Heidelberg EC manual](https://www.heidelberg-wallbox.eu/media/pdf/0e/2c/57/Heidelberg-Wallbox-EnergyControl-Bedienung-DEU.pdf) (chapter 1.6).

The [installation manual](https://www.heidelberg-wallbox.eu/media/pdf/33/1f/4e/Heidelberg-Wallbox-EnergyControl-Montage-DEU.pdf) has all the relevant info on setting the dip switches etc. Just in case you want to double/tripple check.

Regarding heidelbridge:
- Please make sure you have built and flashed using the "heidelberg" environment, not "dummy". 
- Change "LOGGING_LEVEL_ERROR" in the platformio.ini to "LOGGING_LEVEL_DEBUG", and build/flash the firmware. Then check the serial monitor and see what you get (is Modbus working properly?)

Feel free to attach an example log here (don't forget to press reset on the ESP32, so that the log also contains the initialization part).

nobsch, 2024-11-01:
Thank you Boris. Error Code is the communication failure (6x White, 3xBlue). I have checked the dip Switches several times. Because there are only 2 changes to do, it is not so complex.
The heidelbridge is compiled with heidelberg environment.
Because the heidelbridge is communicating i am not convinced that the log files will help. I will try to get information over the serial interface. But it will last a little bit

alexseuf, 2024-11-23:
@nobsch did you conect L and N the right way ( even if you have got a test setup with schuko connector). because if you connect L and N mixed up, the heidelberg box is blinking the same error code (=input voltage error) as with a communicating error...

nobsch, 2024-11-23:
You mean the Power connection of the wallbox?
This is OK. The blinking code is only when i switch to modbus.
Without modbus configuration the Wallbox works for 7 months without problems

sja, 2024-12-27:
258 is listed [here](https://github.com/BorisBrock/HeidelBridge/blob/main/src/Configuration/Constants.h#L33), and is set to `4` [here](https://github.com/BorisBrock/HeidelBridge/blob/main/src/Components/Wallbox/HeidelbergWallbox.cpp#L23).
257 is not defined, I think its not set

BorisBrock, 2024-12-27:
I just published version 1.8.0, which will set both registers on startup:

- 257 (watchdog timeout) is set to 0 -> Watchdog disabled
- 258 (standby mode) is set to 4 -> Standby disabled

On startup, the Heidelbridge should  print debug messages indicating which defaults are being written. In case the values can not be set, proper error messages will be printed.

Both defaults can be changed in Constants.h, if required.
Why the Heidelberg support suggested setting 258 is not clear to me. Disabling the standby seems more appropriate here... But feel free to try it with both values and see which ones work best.

Thank's for testing :+1:

commit 3d89af11 (2025-01-26): Build fix #2
closed 2024-12-29 by BorisBrock