BorisBrock/HeidelBridge #8: Enhancement: WiFi manager instead of static credentials

Giermann, 2025-01-17:
Although a charging point does not change the WiFi network too often, it would enable to publish ready-to-use-binaries, if you add a WiFi credentials manager, i.e. [NetWizard](https://github.com/ayushsharma82/NetWizard).
Further, an OTA ability like [ElegantOTA](https://github.com/ayushsharma82/ElegantOTA) would help to update the ESP32 in the future.

BorisBrock, 2025-01-18:
Hi, both ideas make sense. I already evaluated several credential managers, but none of them did work good enough. NetWizard for example is very limited (no custom name, no custom logo, ads for the pro version, restrictive licensing).
Maybe I'll write my own onboarding library (could be useful for other projects as well)...

BorisBrock, 2025-01-29:
These features are now being worked on. But there is no ETA yet.

closed 2025-01-29 by BorisBrock