BorisBrock/HeidelBridge #10: Compilation error v.1.9

Ledikie, 2025-01-25:
Beim compilieren der Version 1.9 bekomme ich folgenden Fehler:
Die Vorgänger Versionen konnte ich immer ohne Fehler compilieren.

Processing heidelberg (platform: espressif32@^6.8.1; board: esp32doit-devkit-v1; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.8.1) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 2.0.0
|-- eModbus @ 1.7.2
|-- AsyncMqttClient-esphome @ 2.1.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\heidelberg\src\Components\AsyncDelay\AsyncDelay.cpp.o
Compiling .pio\build\heidelberg\src\Components\Logger\Logger.cpp.o
Compiling .pio\build\heidelberg\src\Components\MQTT\MQTTManager.cpp.o
Compiling .pio\build\heidelberg\src\Components\Modbus\ModbusRTU.cpp.o
Compiling .pio\build\heidelberg\src\Components\Modbus\ModbusTCP.cpp.o
Compiling .pio\build\heidelberg\src\Components\Statistics\Statistics.cpp.o
Compiling .pio\build\heidelberg\src\Components\Wallbox\DummyWallbox.cpp.o
Compiling .pio\build\heidelberg\src\Components\Wallbox\HeidelbergWallbox.cpp.o
Compiling .pio\build\heidelberg\src\Components\WiFi\WifiConnection.cpp.o
Compiling .pio\build\heidelberg\src\Configuration\Credentials.cpp.o
Compiling .pio\build\heidelberg\src\Main.cpp.o
Building .pio\build\heidelberg\bootloader.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generating partitions .pio\build\heidelberg\partitions.bin
Compiling .pio\build\heidelberg\libe20\SPI\SPI.cpp.o
Compiling .pio\build\heidelberg\lib7b9\AsyncTCP\AsyncTCP.cpp.o
Compiling .pio\build\heidelberg\liba51\ESPAsyncTCP\AsyncPrinter.cpp.o
Compiling .pio\build\heidelberg\liba51\ESPAsyncTCP\ESPAsyncTCP.cpp.o
Compiling .pio\build\heidelberg\liba51\ESPAsyncTCP\ESPAsyncTCPbuffer.cpp.o
Compiling .pio\build\heidelberg\liba51\ESPAsyncTCP\SyncClient.cpp.o
Archiving .pio\build\heidelberg\libe20\libSPI.a
Compiling .pio\build\heidelberg\liba51\ESPAsyncTCP\tcp_axtls.c.o
.pio/libdeps/heidelberg/ESPAsyncTCP/src/ESPAsyncTCPbuffer.cpp:27:10: fatal error: debug.h: No such file or directory

***************************************************************
* Looking for debug.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:debug.h"
* Web  > https://registry.platformio.org/search?q=header:debug.h
*
***************************************************************

BorisBrock, 2025-01-25:
Hi Ledikie,
mit Version 1.8.0 hatte ich Folgendes an der Platformio.ini geändert:
https://github.com/BorisBrock/HeidelBridge/commit/0cc3ecebd8e032949f82f32bf233bef695f08f82

Das kannst du ja mal probehalber zurücknehmen, und schauen ob es damit klappt. Andere relevante Änderungen in letzter Zeit hat es eigentlich nicht gegeben...

Ledikie, 2025-01-26:
Hi Boris
die 1.8.0 funktionierte bei mir einwandfrei.

Nur die 1.9.0 bricht mit Fehler ab

BorisBrock, 2025-01-26:
Ahoi,
ich habe ein bisschen was korrigiert. Probier es mal mit der neuen Version 1.9.1.

Ledikie, 2025-01-27:
Helau!
Mit der 1.9.1 gibt's keinen Fehler mehr.
Compilieren funzt wie früher.

closed 2025-01-27 by Ledikie