BorisBrock/HeidelBridge #11: How to reset all values

wunderbaum, 2025-01-27:
Hello,
I don't know how, but since I updated to 1.9.1 I have strange values in the mqtt-explorer:

For example:
`heidelbridge/energy_meter = 251658.25
temperature = 230.40
`

BorisBrock, 2025-01-27:
Hi Wunderbaum,
currently I'm unable to reproduce the issue you are describing. Could set the logging level `LOGGING_LEVEL_DEBUG` and share the outputs here?

By the way:
The energy meter value is not stored in the HeidelBridge. It is read directly from the Heidelberg wallbox.

wunderbaum, 2025-01-27:
Many thanks!
I investigated a little bit more as I still have my wallbox on my workbench. Maybe those values where produced while I connected vice versa L/N crossed?

Is it possible to reset the energy meter in the box as I have not even started?

BorisBrock, 2025-01-28:
Hi Wunderbaum,
I don't think that's possible.  This is straight from the wallbox manual:

> Electrical energy drawn by the vehicles since commissioning of the wallbox. The
> register content is not lost when the wallbox is disconnected from the mains. A
> reset is not possible

So I guess you'll have to live with the offset (or fork this repo and offset subtraction to the code) 😄

closed 2025-01-28 by BorisBrock

wunderbaum, 2025-01-28:
Luckily, that was not necessary.

As I am trying to get the code running on a ES32-Super-Mini I think I mixed up one of the compiler-options. Now I started from scratch with it and all values are as they has to be.
Now I hopefully have the right settings.