BorisBrock/HeidelBridge #33: No reconnect to Wifi Network

draccor, 2025-07-25:
If wifi router breaks down or reboots the heidelbridge loses the connection. Thats normal. But after wifi is back again the heidelbridge does not reconnect to the hotspot again. After power off the esp32 and power on again the connection is established.

BorisBrock, 2025-07-26:
The Heidelbridge has a reconnect mechanism, that should restore the WiFi connection. If that's not working for you, could you please provide a log?

-> Build Heidelbridge with logging set to debug, connect via USB, turn WiFi router off/on

draccor, 2025-07-26:
Moin,

 

ich werde die Tage mal einen zweiten ESP bespielen und gucken, ob sich dieser genauso verhält.

 

Es ist für mich etwas aufwendig die Loggings zu bekommen, weil ich den ESP in die Heidelberg Wallbox eingebaut habe.

 

Ich melde mich erneut.

 

Vielen Dank 😊

BorisBrock, 2026-02-19:
Stale issue -> close